The-OpenROAD-Project/OpenROAD-flow #18: Add

stdavids, 2020-01-08:
I tried getting the tools running on CentOS 7 and 8 both of which had some issues with software dependencies with the base upstream software repositories.

Cent OS 7 issues
=================
1. Glibc was too old (had to compile a newer version of glibc)
2. Libffi was too new (fixed by renaming libffi.so.6 to libffi.so.5 using a symlink and adding that path to the LD_LIBRARY_PATH)

CentOS 8 issues
==================
1. Libffi was too new (same as CentOS 7)
2. Libtcl was too new (similar to libffi, fixed by renaming libtcl8.6.so to libtcl8.5.so using a symlink and added the path to the LD_LIBRARY_PATH)

It would be good to get some instructions for how to get the flow running on a fresh install of an officially support operating system.

rovinski, 2020-01-08:
Thanks for the heads-up. This was probably a mistake on my part in building the release binaries using a  local modules instead of inside Docker (such that the lib versions are tied to CentOS7). I will try to fix and re-release.

tajayi, 2020-01-09:
Detailed instructions have been setup to build/install all binaries on CentOS 6/7 here: bit.ly/openroadJan2020

rovinski, 2020-01-09:
The prebuilt binaries have been updated such that everything was built in a CentOS 7 container. Hopefully this should alleviate the issue.

closed 2020-01-09 by rovinski

stdavids, 2020-01-09:
New uploaded binaries are working out of the box on CentOS7.