The-OpenROAD-Project/OpenROAD-flow #21: Missing PORT9.dat file

tunghoang290780, 2020-01-22:
This PORT9.dat file must be copied into OpenROAD-flow/OpenROAD/etc similar to other POWV9.dat, POST9.dat. The PORT9.dat belongs to the yosys package

tajayi, 2020-01-22:
I am aware that the  PORT9.dat file is not included in the OpenROAD build. I believe it only requires FluteLUT files: POWV9.dat and POST9.dat.

Can you please elaborate on any specific issue you are experiencing? Did you get an error while performing some operation?

tunghoang290780, 2020-01-22:
So far, I do not see the problem with the missing file PORT9.dat. If that file is no longer use, it is possible to clean up Makefile or script that refer to the PORT9 file

tajayi, 2020-01-22:
I assume you are referring to https://github.com/The-OpenROAD-Project/OpenROAD-flow/blob/master/flow/Makefile#L126?

It was used once upon a time ago but didn't get cleaned up. I have deleted references to that file in a local commit. It'll take some time before being pushed.

Thanks

tunghoang290780, 2020-01-22:
Hi @tajayi 

That is correct! Thanks

rovinski, 2020-02-06:
*.dat files are no longer required as of 048a503. This issue should be resolved now.

closed 2020-02-06 by rovinski