The-OpenROAD-Project/OpenROAD-flow #22: How to create modified macro LEF file?

tunghoang290780, 2020-01-22:
Hi,

I am trying to build a platform for specific in-house standard cell libraries using OpenROAD. One of requirement is to generate merged.lef from NangateOpenCellLibrary.tech.lef and modified macro NangateOpenCellLibrary.macro.mod.lef file. I observe there are differences between the original file ./platforms/nangate45/NangateOpenCellLibrary.macro.lef and the modified NangateOpenCellLibrary.macro.mod.lef (case sensitive, field order, and many ANTENNAPARTIALMETALAREA are added per macrocell)

Is there any guideline to generate NangateOpenCellLibrary.macro.lef manually or automatically (via util script)? 

Thanks

tajayi, 2020-01-23:
The primary reason for modifying the NangateOpenCellLibrary.macro.mod.lef is because some tools we used early on during development did not support "POLYGON" pin shapes in the macros. I am uncertain if this is still the case. Note that we have been able to add several other kits without modifications.

NangateOpenCellLibrary.macro.mod.lef was generated using from the gds using commercial tools to generate an abstract lef.

rovinski, 2020-02-06:
This is still the case at the moment. TritonRoute does not appear to support `POLYGON` pins.
```
Error: genGuides_gCell2TermMap unsupoprted pinfig
```
`RECT` pins are supported.

`ANTENNA*` constructs will not cause issues and do not need to be removed. The fact that `NangateOpenCellLibrary.macro.mod.lef` has those statements removed is just a byproduct of re-abstracting the cells without calculating antenna area.

tunghoang290780, 2020-02-07:
> NangateOpenCellLibrary.macro.mod.lef was generated using from the gds using commercial tools to generate an abstract lef.
@tajayi @rovinski    I am guessing that to generate NangateOpenCellLibrary.macro.mod.lef, I need to import NangateOpenCellLibrary.gds into commercial tool (i.e. IC Virtuoso) then export another abstract LEF (e.g. NangateOpenCellLibrary.macro.mod.lef)

tajayi, 2020-02-07:
That's one way. It's also theoretically possible to write-up some sort of python (or language of choice) script to convert polygon shapes in the lef to a series of rectangles

rbarzic, 2020-03-21:
Has anybody started writing such a script ? I may consider writing one myself but if someone has already started I'll likely help

tajayi, 2020-03-21:
I'm not aware of any script and/or efforts to write one. 
I have also been lucky enough to work with several kits and this is the first one I'm seeing that uses polygons instead of rectangles.