The-OpenROAD-Project/OpenROAD-flow #32: GCD failed with OpenROAD-flow

tunghoang290780, 2020-02-25:
I have pulled the latest code for OpenROAD (commit c89bf32) flow and re-run GCD (my sanity test). Apparently, the flow failed with the GCD design and below error appeared

```
##Power Delivery Network Generator: ./platforms/nangate45/pdn.cfg 1
Design Name is gcd
Reading BEOL LEF and gathering information ...
ERROR: Layer specified for std. cell rails not in list of layers.
**ERROR: Layer specified for std. cell rails not in list of layers.**
Makefile:221: recipe for target 'results/nangate45/gcd/2_5_floorplan_pdn.def' failed
make: *** [results/nangate45/gcd/2_5_floorplan_pdn.def] Error 1
```

I keep running flow and it achieves further step, till to another error

> Final O/bin/bash: line 1: 14336 Segmentation fault      (core dumped) openroad -no_init ./scripts/global_route.tcl 2>&1
>      14337 Done                    | tee ./logs/nangate45/gcd/5_1_fastroute.log
> Makefile:347: recipe for target 'results/nangate45/gcd/route.guide' failed
> make: *** [results/nangate45/gcd/route.guide] Error 139
>

rovinski, 2020-02-25:
Hi @tunghoang290780,

We now have an issue template for reporting bugs. If it did not show up as an option when filing the issue, please let me know.

This is one of the designs we use for our regression tests. I have to imagine that this is an issue of an old tool installation.

Was this build from a fresh clone of OpenROAD-flow? If not, you may want to first try

1. `git submodule update --init --recursive`
2. delete `tools/build`
3. Rebuild the tools (`build_openroad.sh`)

Otherwise, just try following the README instructions to perform a fresh clone (options 2 or 3). Using the build exports is not recommended at this time as they are out of date. We have automatic uploads of binaries on our roadmap but there is no ETA.

tunghoang290780, 2020-02-27:
@rovinski I can build docker successfully but the docker is not run. Do you have any idea?

```
$ dk_li 
REPOSITORY             TAG                 IMAGE ID            CREATED             SIZE
openroad/flow          latest              fa8731afdacc        8 minutes ago       1.26GB
openroad               latest              93af20dda1b0        11 minutes ago      1.81GB
openroad/tritonroute   latest              0c99082787c0        18 minutes ago      385MB
openroad/yosys         latest              0a532a7838be        24 minutes ago      2.42GB
centos                 centos7             5e35e350aded        3 months ago        203MB
$ ./setup_env.sh
$ sudo docker run -u $(id -u ${USER}):$(id -g ${USER}) openroad/flow bash
$ >>> Nothing happen

```

rovinski, 2020-02-27:
That is a bug in the README. You need to use the `-it` flag to run docker interactively:
```
docker run -it -u $(id -u ${USER}):$(id -g ${USER}) openroad/flow bash
```
Otherwise, docker will simply create a container and exit.

I will fix the README for the next update to master.

rovinski, 2020-02-27:
README fixed in f23aee8.

tunghoang290780, 2020-02-27:
@rovinski thanks for fixes. Meanwhile, I have successfully built the OpenROAD flow from the source (local build) and it works very well.

closed 2020-02-27 by tunghoang290780