The-OpenROAD-Project/OpenROAD-flow #35: Docker container won't run

VictorySpecificationII, 2020-03-26:
Hi, 

Thank you for these tools.

I have a problem; typing 

sudo docker run -u $(id -u ${USER}):$(id -g ${USER}) openroad/flow bash

does not open a shell for me; What am I missing?

Many thanks

rovinski, 2020-03-26:
Hi @VictorySpecificationII,
Please see https://github.com/The-OpenROAD-Project/OpenROAD-flow/issues/32#issuecomment-591732171. It has not made its way to master yet.

closed 2020-03-26 by rovinski

VictorySpecificationII, 2020-03-26:
Hi again, 

The container is working fine now. CD'ing into the flow directory within openroad/flow and executing make outputs:

Writing replaced file: objects/nangate45/gcd/merged.lib
mkdir -p ./results/nangate45/gcd ./logs/nangate45/gcd ./reports/nangate45/gcd
yosys -l ./logs/nangate45/gcd/1_1_yosys.log -c scripts/synth.tcl
/bin/bash: yosys: command not found
make: *** [results/nangate45/gcd/1_1_yosys.v] Error 127

Is openroad/flow somehow connected with openroad/yosys? Is there a procedure to connect them in docker? Yosys seems to be in the container.

Thanks for putting up with me.

rovinski, 2020-03-27:
Hi @VictorySpecificationII,

Once you start the shell inside docker, you need to make sure that your environment is set up. This can be done with:

```
source setup_env.sh
```