The-OpenROAD-Project/OpenROAD-flow #41: Various minor problems when trying to take OpenROAD-flow for a spin

oharboe, 2020-04-04:
1. I had to change the launch command from: ```docker run -u $(id -u ${USER}):$(id -g ${USER}) openroad/flow bash``` to ```docker run -ti -u $(id -u ${USER}):$(id -g ${USER}) openroad/flow bash```, i.e. add the ```-ti`` option
2. The harness design didn't work with the attached RotateLeft.v ```make DESIGN_CONFIG=designs/harness.mk DESIGN_NAME=RotateLeft```. Error message below.

```
$ make DESIGN_CONFIG=designs/harness.mk DESIGN_NAME=RotateLeft
[deleted]
==========================================================================
report_design_area
--------------------------------------------------------------------------
Design area 4398 u^2 0% utilization.
Error: 1_synth.sdc, 10 -name or port_pin_list must be specified.
make: *** [results/nangate45/RotateLeft/2_1_floorplan.def] Error 1
bash-4.2$ 
```

oharboe, 2020-04-04:
[RotateLeft.zip](https://github.com/The-OpenROAD-Project/OpenROAD-flow/files/4430946/RotateLeft.zip)

oharboe, 2020-04-04:
Is there an automated pre-merge test the designs/src/harness?

rovinski, 2020-07-07:
> i.e. add the -ti option

Fixed in f23aee8

> The `. setup_env.sh` is missing from the instructions.

Fixed in 8a1c931

> Below is how I ran a quick test on a small module. Initially I picked a module without a clock, which didn't work, but the error message was OK if not great.

As of f25d8b0 TritonCTS should now gracefully exit if there are no clock trees in the design, which allows the design to continue through the flow. TritonCTS will still report an error about no clock nets detected in the log.

> Perhaps the current working folder could be mounted instead of having the flow folder be part of the Docker image

A user is free to do so if they desire. I made a note of it in 7b0a77f.

I believe all points in this issue have been addressed so this issue will be closed now.

closed 2020-07-07 by rovinski