The-OpenROAD-Project/OpenROAD-flow #44: LRU.v takes a very long time to go through the OpenROAD-flow

oharboe, 2020-04-07:
**Describe the bug**

The tiny LRU.v should go through the OpenROAD-flow in a reasonable amount of time, say 15 minutes.

To reproduce:

1. Check out #43 
2. Run ```./build_openroad.sh```
2. Extract "LRU.v", attached to a folder.
3. Run the openroad smoketest. I don't know how long it takes to complete. I stopped it after ca. 1 hr.

```
$ cd OpenROAD-flow
$ time ./openroad-smoketest <path to LRU.v> LRU
[output from OpenROAD-flow with the harness project deleted]
INFO] Nesterov: 190 OverFlow: 0.5836 ScaledHpwl: 293287.4375
[INFO] Nesterov: 200 OverFlow: 0.4756 ScaledHpwl: 293824.2188
[ctrl-c]
```

**Expected behavior**

The flow should complete in less than 15 minutes.

**Environment (please complete the following information):**

Check out #43 

**File Uploads**

LRU.v attached in .zip format.
[LRU.zip](https://github.com/The-OpenROAD-Project/OpenROAD-flow/files/4443602/LRU.zip)

rovinski, 2020-04-11:
The problem is that the core area is set to an unreasonably large size: https://github.com/The-OpenROAD-Project/OpenROAD-flow/blob/229a1f41fc9bae5f5dfd871c799edceb985d4b7e/flow/designs/harness.mk#L14-L15

Many operations are proportional to the core area so this will definitely take a long time. You should reduce the core area to a reasonable size or use automatic sizing based on utilization, e.g.
```
# Core utilization in %
export CORE_UTILIZATION = 10.0
# Core height / core width
export CORE_ASPECT_RATIO = 1.0
# Core margin in um
export CORE_MARGIN = 2.0
```

Using the above modification, your design runs in about 15 seconds for me.

oharboe, 2020-04-11:
@rovinski  Fantastic! :-) Works like a charm. I created #49 to fix this. I'm trying it with the openroad branch and https://github.com/The-OpenROAD-Project/OpenROAD/issues/197

rovinski, 2020-04-14:
@oharboe I merged it into the openroad branch which gets merged into master periodically. We want to keep a linear history so please make all pull requests to the openroad branch, thanks.

oharboe, 2020-04-14:
@rovinski Thanks!

Riding my hobbyhorse: we are using Mergify to automatically rebase and merge after successful automatic pull request tests
 
We use Jenkins as well as Google Cloud Build.

We use a google cloud datatbase to keep track of what parts of the code that need to be retested using a content, not git, hash. This yields pull request build times ranging form minutes to hours.

We use a Mergify "strict" policy so pull requests are rebased and rebuilt  automatically if needed(mergify has smarts to queue to avoid too much rebuilding).

Pull requests are always directly to master for our monorepository using git subtree.

oharboe, 2020-04-14:
This particular issue is solved for me. 

The larger issue of how a user is to guess and learn how to adjust hundreds of options with, effectively, infinitely many combinations and many hours turnaround times remains unsolved....

rovinski, 2020-04-14:
@oharboe unfortunately our sponsor has prioritized improving the feature set and quality of results over usability in the short term. We have all of our bandwidth dedicated to supporting a full SoC design in advanced nodes (i.e. 16/14nm). We have a very long wish list of usability and infrastructure improvements, but it doesn't look like anyone is going to be working on them until at least July.

The notes on merge strategy, etc. are duly noted.

oharboe, 2020-04-15:
@rovinski Got it. Thanks for lending an ear. 

I don't know what the procedure is w.r.t. closing issues, who does that?

PR #49 (2020-04-14): harness: automatically pick utilization and die area
Avoid pathological running times for tiny smoketest designs. Fixes #44
closed 2020-04-16 by rovinski