The-OpenROAD-Project/OpenROAD-flow #45: Small circuits

RatkoFri, 2020-04-07:
Greetings,

I am trying to synthesize small circuits like multipliers. In one tool (QFLOW) my designs exhibit really small energy consumption. On the other hand OpenROAD, I get totally different results. Can you give some tips on how to efficiently synthesize small designs?

Ratko

rovinski, 2020-04-08:
Hi @RatkoFri,

Can you please be more specific about the behavior you are seeing? Can you provide us with the verilog and design configuration inputs? Does the difference show up when targeting the nangate45 library, and if so, can you provide the specific numbers you are seeing?

RatkoFri, 2020-04-08:
Yeah sure. You will find everything you need in an attachment. Concerning results, I think that tool gives a bigger delay than actual. During development, I especially focused on optimizing the design.
[mult.zip](https://github.com/The-OpenROAD-Project/OpenROAD-flow/files/4448932/mult.zip)

RatkoFri, 2020-04-09:
I figured the problem. In one part of my design, I tried to implement the NOR gate with this construction:

assign tmp50 = ~ (data_i[5:4] | data_i[7:6]);

I presume that flow does not recognize that I want to use the NOR gate. Instead, it probably uses an OR gate and inverter to synthesize the NOR operation.

closed 2020-04-09 by RatkoFri

rovinski, 2020-04-09:
@RatkoFri If the statement `assign tmp50 = ~ (data_i[5:4] | data_i[7:6]);` is being synthesized differently than what you are expecting, then that would be a synthesis issue from yosys. Both Qflow and OpenROAD use yosys, so I don't see how that could be the origin of the problem unless they are different versions.

> it probably uses an OR gate and inverter to synthesize the NOR operation

You can check the netlist file `1_1_yosys.v` to see exactly which gates it was synthesized to.