The-OpenROAD-Project/OpenROAD-flow #50: Add example project that uses fake RAM for attached Verilog used to infer RAM

oharboe, 2020-04-11:
**Preamble (delete this section after reading)**

The attached DualPortMaskedWrite.sv file is used to infer BRAM in Quartus.

I'd like to see an example design in OpenROAD-flow/designs on how RAM can be mocked/faked so as to be able to run design throw the OpenROAD-flow.

[DualPortMaskedWrite.zip](https://github.com/The-OpenROAD-Project/OpenROAD-flow/files/4464415/DualPortMaskedWrite.zip)

tajayi, 2020-04-11:
This is an example design that uses bsg_fakeram in Nangate45
https://github.com/The-OpenROAD-Project/OpenROAD-flow/blob/master/flow/designs/nangate45/bp_fe_top.mk

The config file used to generate those in the bsg_fakerame tool is here
https://github.com/The-OpenROAD-Project/OpenROAD-flow/blob/master/flow/designs/src/bp_be_top/fakeram.cfg

oharboe, 2020-04-11:
@oharboe Thanks! Dual port SRAM is idiomatic for FPGA, so any designs coming from FPGA are likely to use them. If I have dual port SRAM that's as wide and deep, 32 bit wide and 512 rows deep, then I can easily modify my design to instantiate them. I don't think I can synthesize dual port RAM based on two single port RAM, right?

I don't particularly care about design size at this point, I just want to see if it is at all possible to synthesize my design as a whole.

tajayi, 2020-04-11:
> I don't think I can synthesize dual port RAM based on two single port RAM, right?
Not "true" dual-port (allows for reads/writes on both ports which are on different clock domains) which are common primitives on FPGAs. If you are familiar with how the rams are used, you might be able to get away with a single port ram. e.g.  If you don't need  2 clock domains and familiar with the logic, you might realize that there are no situations where you are doing 2 writes in the same cycle or it can be staggered. etc.. Unfortunately, this may be intrusive to your design

If you can live with dual read and single write port, you can add a layer that replicates the sram (twice the area). Write both in the same cycle and read separately.

We can also nicely ask @stdavids nicely for dual-port ram support ;)

oharboe, 2020-04-11:
Quite.... :-) There are some other problems that have to be solved first, but it looks like dual port RAM might be available by then...