The-OpenROAD-Project/OpenROAD-flow #51: Design get stuck in flow, with no output to indicate what's wrong

oharboe, 2020-04-11:
I've have a couple of designs that pass in a few minutes through the "harness" project.

However, when I try a bigger project, the flow gets stuck, without any indication as to whether there's any progress or it's stuck.

The design quickly (10 minutes?) gets to where it's apparently stuck below.

This is using "openroad-smoketest" script from #43 

```
[deleted]
Notice 0: Design: BlockDecompressionUnit
Notice 0: 		Created 100000 Insts
Notice 0: 		Created 200000 Insts
Notice 0: 		Created 300000 Insts
Notice 0: 		Created 400000 Insts
Notice 0: 		Created 500000 Insts
Notice 0: 		Created 600000 Insts
Notice 0: 		Created 700000 Insts
Notice 0: 		Created 800000 Insts
Notice 0:     Created 6677 pins.
Notice 0:     Created 809747 components and 2193785 component-terminals.
Notice 0:     Created 5 special nets and 1619494 connections.
Notice 0:     Created 164782 nets and 574291 connections.
Notice 0: Finished DEF file: ./results/nangate45/BlockDecompressionUnit/4_cts.def
Adjust layer 2 in 50.0%
Adjust layer 3 in 50.0%
[apparently stuck here]
```

oharboe, 2020-04-11:
eventually, the flow went on...

```
[continuing...]
 *****************
 *   FastRoute   *
 *****************

 > Params:
 > ---- Min routing layer: 2
 > ---- Max routing layer: 10
 > ---- Global adjustment: 0.15
[...]
```

oharboe, 2020-04-11:
After a few hours, the flow seems stuck here. It's literally stuck after writing out the "t" letter...

[image]

oharboe, 2020-04-12:
Tried it again, stuck at exactly the same place, "updatet", then nothing more...

oharboe, 2020-04-12:
I tried a much scaled down version of this design and it went through in a few minutes, producing a GDS file.

tajayi, 2020-04-12:
Sorry, haven't had too much time to look into this issue. 

> Notice 0:     Created 809747 components and 2193785 component-terminals.

My guess is that there might be too many instances for use to handle. What's the cell count from yosys?

oharboe, 2020-04-12:
@tajayi  171301 cells from yosys

oharboe, 2020-04-12:
I've got a design that's 130943 cells in Yosys. It completes in 6480 seconds, producing a GDS file. It's 5248 Altera ALMs.

tajayi, 2020-04-12:
> @tajayi 171301 cells from yosys

So the step failing is "global route". For some reason, it's finding it challenging or impossible to create route guides. This could be due to congestion in the design, bad placement or a number of other reasons.

My recommendation is to take a look at `4_1_cts_prefillcell.def` in layout. Try to check for any visible issues (feel free to post the screenshot).

To relieve some congestion issue you can tweak place density. This affects how densely cells are placed together. See an example of a design overwriting the default here:
https://github.com/The-OpenROAD-Project/OpenROAD-flow/blob/openroad/flow/designs/nangate45/bp_be_top.mk#L19

The default is set here:
https://github.com/The-OpenROAD-Project/OpenROAD-flow/blob/openroad/flow/platforms/nangate45/config.mk#L82

oharboe, 2020-04-12:
@tajayi  Thanks! I'll try reducing placement density. Silly question what do you mean by "take a look at  4_1_cts_prefillcell.def in layout"? Is there a tool?

tajayi, 2020-04-12:
I meant to say "klayout", but auto-correct...
Yes, klayout (which is used to generate the gds) can be used to inspect the design.

You should be able to run `make klayout_4_1_cts_prefillcell.def` to trigger a recipe that opens it up in the app.

oharboe, 2020-04-13:
Reduce placement density and scaled down the design, it still get stuck in the same spot. I'm going to have to wait to arrange for you to get the Verilog here, to make progress, I think.