The-OpenROAD-Project/OpenROAD-flow #67: Pathologically large negative slack with small design with RAM block

oharboe, 2020-05-06:
@rovinski @tutu I've managed to reduce this pathologically large negative slack down to ca. 100 lines of simple Verilog(aptos2x2/StateColumn.v).

I've uploaded the confidential files as confidential-statecolumn-very-large-negative-slack.tar.bz2 (includes everything all the way to the .gds file).

If this proves to be a useful test-case and we can make it a bit more readable, I think we can upload it as a publicly available example.

```
==========================================================================
report_checks
--------------------------------------------------------------------------
Startpoint: _444_ (rising edge-triggered flip-flop clocked by clock)
Endpoint: memory.my_ram.u_fakeram45_1r1w_1024x64
          (rising edge-triggered flip-flop clocked by clock)
Path Group: clock
Path Type: max

  Delay    Time   Description
---------------------------------------------------------
   0.00    0.00   clock clock (rise edge)
   0.00    0.00   clock network delay (ideal)
   0.00    0.00 ^ _444_/CK (DFF_X1)
   0.08    0.08 ^ _444_/Q (DFF_X1)
   0.06    0.14 ^ _192_/Z (BUF_X1)
   0.09    0.24 ^ _217_/ZN (AND4_X1)
   0.03    0.27 ^ _225_/ZN (OR2_X1)
   0.03    0.30 v _226_/ZN (NAND3_X1)
   0.03    0.33 ^ _373_/ZN (OAI21_X1)
   0.02    0.34 v _374_/ZN (AOI21_X1)
5194.03 5194.37 ^ _376_/ZN (AOI21_X1)
   0.03 5194.40 ^ memory.my_ram.u_fakeram45_1r1w_1024x64/addr_in[4] (fakeram45_1r1w_1024x64)
        5194.40   data arrival time

   4.00    4.00   clock clock (rise edge)
   0.00    4.00   clock network delay (ideal)
   0.00    4.00   clock reconvergence pessimism
           4.00 ^ memory.my_ram.u_fakeram45_1r1w_1024x64/clk (fakeram45_1r1w_1024x64)
  -0.43    3.57   library setup time
           3.57   data required time
---------------------------------------------------------
           3.57   data required time
        -5194.40   data arrival time
---------------------------------------------------------
        -5190.83   slack (VIOLATED)

```

tajayi, 2020-05-06:
The issue is probably due to the fakerams being a bit too fake. The timing data (*.lib) associated with the fakeram is probably ludicrous or using the wrong units somewhere (eg. ps vs ns).

oharboe, 2020-05-06:
@tajayi  If I add a register on the input address, then I get terrible(12ns) but not pathological (thousands ns) negative slack.

tajayi, 2020-05-06:
I have validated that the timing reports are sane for real process nodes with real srams.  The issue is probably caused by:

1. Inconsistent units in the *.lib file for the standard cells and srams. (Specifically leakage_power_unit and capacitive_load_unit). The merge process (used for yosys) doesn't attempt to make the units coherent.

2. It's possible that they are wild numbers in the lib files from bsg_fakeram (or cacti).

3. Yosys-abc doesn't do a good job of buffering/resizing. The timing numbers may improve after the resizing / design fixing step