The-OpenROAD-Project/OpenROAD-flow #79: [Build] Dockerfile build fails due to legacy ius-release.rpm links

ilesser, 2020-05-30:
**Expected behavior**
Dockerfile should build without errors.

**Screenshots**
Error after running `./build_openroad.sh`
[image]

**Environment:**
Please only report issues for supported OSs.
 - OS: Linux Mint 19.1 Tessa [CentOS 7 in docker]
 - OpenROAD-flow commit [229a1f41]

ilesser, 2020-05-30:
Changing the links to `https://repo.ius.io/ius-release-el7.rpm` should fix everything.

I can provide a PR during the weekend.

ilesser, 2020-05-30:
Updating the links fixed the first error but now there is another error since it can't find `git2u`:
[image]

zhanghb97, 2020-06-02:
> Updating the links fixed the first error but now there is another error since it can't find `git2u`:
> [image]

You can find the available git version with `repoquery --whatprovides git`.

As for me, I can find the following versions:
```
repoquery --whatprovides git 
git-0:1.8.3.1-21.el7_7.x86_64 
git-0:1.8.3.1-22.el7_8.x86_64 
git222-0:2.22.2-1.el7.ius.x86_64 
git222-0:2.22.3-1.el7.ius.x86_64 
git224-0:2.24.2-1.el7.ius.x86_64
```

I solve the problem by using git224 (modify the Dockerfile `./tools/OpenROAD/Dockerfile`).

rovinski, 2020-06-02:
Fixed in ed2ecee.

closed 2020-06-02 by rovinski