The-OpenROAD-Project/OpenROAD-flow #82: The output of cmd [write_verilog] lacks wire information

puckbee, 2020-06-05:
I tried OpenRoad flow and want to output the netlist in each step. So that I could feed it into commercial tool [ Synopsys Formality ] to verify the correctness. 

However, the result I got from the cmd [write_verilog] lacks the [wire] part. I traced the code and found out the implementation of [write_verilog] in OpenSTA/verilog/VerilogWriter.cc 

Is it a bug in writeModule() that someone forgets to output the [wire] info?

void
writeVerilog(const char *filename,
             bool sort,
             Network *network)
{
  if (network->topInstance()) {
    FILE *stream = fopen(filename, "w");
    if (stream) {
      VerilogWriter writer(filename, sort, stream, network);
      writer.writeModule(network->topInstance());
      fclose(stream);
    }   
    else
      throw FileNotWritable(filename);
  }
}

void
VerilogWriter::writeModule(Instance *inst)
{
  Cell *cell = network_->cell(inst);
  fprintf(stream_, "module %s (",
          network_->name(cell));
  writePorts(cell);
  writePortDcls(cell);          ------------------------------------------------------------
  fprintf(stream_, "\n");       ----------------    Here   ---------------------------------
  writeChildren(inst);        ----------------------------------------------------------------
  fprintf(stream_, "endmodule\n");
  written_cells_.insert(cell);

  for (auto child : pending_children_) {
    Cell *child_cell = network_->cell(child);
    if (!written_cells_.hasKey(child_cell))
      writeModule(child);
  }
}

rovinski, 2020-06-07:
Can you provide a simple test case and the actual behavior vs. expected behavior?
Also, we do have bug templates. Please use them to help us resolve the issue more quickly.

puckbee, 2020-06-08:
Um, I didn't find the bug template anywhere. But  I can provide more details that may help.

Test Case: https://github.com/cliffordwolf/picorv32/blob/master/picorv32.v
Config file: Tiny-test.mk with minor modification
Behaviors:  After you finished the flow, try the 4_cts.v, no wires exist in this file. However, there should be wires in the design, which you may confirm in the 1_synth.v

rovinski, 2020-06-08:
When you click on the "New Issue" button, you should get the following screen. Was that not the case?
[image]

Also can you please be more specific about what you mean by "no wires"? As in, no wire declarations?

puckbee, 2020-06-08:
Thank you very much. I will use the template the next time.

yes, correct. There are no wire declarations. Is that a bug or a mis-using of this command?

ahmed-agiza, 2020-06-08:
You can rewrite the netlist using Yosys if you need the wire declarations:
`
read_verilog -noopt netlist.v
write_verilog -noattr -noexpr -nohex -nodec netlist_rewrite.v
`

puckbee, 2020-06-10:
Actually, Yosys works well to me, and I can find the wire declarations in my 1_synth.v. However, the wire declarations seem missing in other netlist files, like 4_cts.v. So that I'm wondering whether it is a bug that the [write_verilog] command in OpenRoad doesn't output the wire declarations.

rovinski, 2020-06-16:
@puckbee Please see previous question. Otherwise, I will go ahead and close the issue as stale.

closed 2020-06-19 by rovinski