The-OpenROAD-Project/OpenROAD-flow #91: Error: global_placement -timing_driven is not a known keyword or flag

promach, 2020-06-24:
For [aes.mk](https://github.com/The-OpenROAD-Project/OpenROAD-flow/blob/master/flow/designs/nangate45/aes.mk) , why do I have the following error during `make DESIGN_CONFIG=./designs/nangate45/aes.mk ` ?

The full terminal log could be [found here](https://github.com/The-OpenROAD-Project/OpenROAD-flow/files/4825808/error.txt)

```
Notice 0: Reading LEF file:  ./platforms/nangate45/lef/NangateOpenCellLibrary.macro.mod.lef
Notice 0:     Created 134 library cells
Notice 0: Finished LEF file:  ./platforms/nangate45/lef/NangateOpenCellLibrary.macro.mod.lef
Notice 0: 
Reading DEF file: ./results/nangate45/aes/2_floorplan.def
Notice 0: Design: aes_cipher_top
Notice 0:     Created 388 pins.
Notice 0:     Created 21324 components and 109458 component-terminals.
Notice 0:     Created 5 special nets and 42648 connections.
Notice 0:     Created 20397 nets and 66810 connections.
Notice 0: Finished DEF file: ./results/nangate45/aes/2_floorplan.def
Error: global_placement -timing_driven is not a known keyword or flag.
0:04.63elapsed 98%CPU 130456memKB
Error: global_placement -timing_driven is not a known keyword or flag.
make: *** [Makefile:247: results/nangate45/aes/3_1_place_gp.def] Error 1
[phung@archlinux flow]$ 
```

rovinski, 2020-06-26:
You have a mismatch between your flow version and tool version. You should run with the version of the tools committed via submodule to the branch of `OpenROAD-flow` you are running.

Also, please use the bug template when filing issues.

closed 2020-06-26 by rovinski

promach, 2020-06-26:
For those interested in this issue, `./build_openroad.sh -o --or-branch openroad` is the correct installation command to use

Besides, remember to `source setup_env.sh`

Here is [an example](https://github.com/promach/noc/tree/development) with detailed installation and usage instructions

rovinski, 2020-06-26:
An explanation: both `OpenROAD-flow` and `OpenROAD` have a branch `openroad`. This is the name of the development branch (don't ask me why). If you are using branch `master` of `OpenROAD-flow`, this will have a submodule tied to the version of `OpenROAD` that is compatible. Branch `openroad` of `OpenROAD-flow` has a much more recent version of the flow which tracks branch `openroad` of `OpenROAD`.

Unless you are a developer, there is generally no reason to use `--or-branch openroad`. You should instead be trying to use the `openroad` branch of `OpenROAD-flow` and install as normal. e.g.

```
git clone https://github.com/The-OpenROAD-Project/OpenROAD-flow.git -b openroad
cd OpenROAD-flow
./build_openroad.sh
```

Do not expect the master branch to be updated frequently until 2H 2020. We are focused mostly on meeting our sponsor-mandated goals.