The-OpenROAD-Project/OpenROAD-flow #104: Broken HEAD of branch openroad

gkamendje, 2020-07-03:
**Describe the bug**
The following build sequence 
```
git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow.git openroad-flow-branch
cd openroad-flow-branch/
./build_openroad.sh -l -p -n
```
fails with a misleading error message: 

```
....
/VerilogLex.cc.o CMakeFiles/OpenSTA.dir/VerilogParse.cc.o
/usr/bin/ranlib tmp_compile/openroad-flow-branch/tools/OpenROAD/src/OpenSTA/app/libOpenSTA.a
make[2]: Leaving directory 'tmp_compile/openroad-flow-branch/tools/build/OpenROAD'
[ 56%] Built target OpenSTA
make[1]: Leaving directory 'tmp_compile/openroad-flow-branch/tools/build/OpenROAD'
Makefile:152: recipe for target 'all' failed
make: *** [all] Error 2
```

The problem here is that building of AntennaChecker sources failed upstream but the compilation flow failed to detect it.  As can be seen below, scrolling backwards through the compilation log reveals the following error message that should had force compilation to stop. But instead, the flow goes on and tries to compile PDNSim sources. 

```
Scanning dependencies of target def
In file included from tmp_compile/openroad-flow-branch/tools/OpenROAD/src/antennachecker/src/MakeAntennaChecker.cc:34:
tmp_compile/openroad-flow-branch/tools/OpenROAD/src/antennachecker/include/antennachecker/AntennaChecker.hh:33:10: fatal error: tcl.h: No such file or directory
 #include <tcl.h>
          ^~~~~~~
compilation terminated.
src/antennachecker/src/CMakeFiles/antennachecker.dir/build.make:95: recipe for target 'src/antennachecker/src/CMakeFiles/antennachecker.dir/MakeAntennaChecker.cc.o' failed
make[2]: *** [src/antennachecker/src/CMakeFiles/antennachecker.dir/MakeAntennaChecker.cc.o] Error 1
make[2]: Leaving directory 'tmp_compile/openroad-flow-branch/tools/build/OpenROAD'
CMakeFiles/Makefile2:2381: recipe for target 'src/antennachecker/src/CMakeFiles/antennachecker.dir/all' failed
make[1]: *** [src/antennachecker/src/CMakeFiles/antennachecker.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: Leaving directory 'tmp_compile/openroad-flow-branch/tools/build/OpenROAD'
make  -f src/OpenDB/src/def/CMakeFiles/def.dir/build.make src/OpenDB/src/def/CMakeFiles/def.dir/build
make[2]: Entering directory 'tmp_compile/openroad-flow-branch/tools/build/OpenROAD'
[ 32%] Building CXX object src/PDNSim/src/CMakeFiles/pdnsim.dir/pdnsim.cpp.o
```
 Editing the two files 
```
 openroad-flow-branch/tools/OpenROAD/src/antennachecker/include/antennachecker/AntennaChecker.hh
 openroad-flow-branch/tools/build/OpenROAD/src/antennachecker/src/AntennaChecker_wrap.cc
```
and replacing  <tcl.h> with <tcl/tcl.h> in the header section fixed the issue.

**Environment (please complete the following information):**
Please only report issues for supported OSs.
 - OS: Ubuntu 18.04 LTS
 - OpenROAD-flow commit 0f69179efb602150b281400f920b405bf3d8917b

**File Uploads**

**Additional context**
My local build_openroad.sh  also has the following compiler option -DCMAKE_CXX_FLAGS=-pthread

rovinski, 2020-07-03:
@gkamendje The "fail upstream" issue is a side effect of building in parallel. It is expected.

With regards to the issue itself, see https://github.com/The-OpenROAD-Project/OpenROAD/pull/363. The fix will be upstreamed here once it is merged.

gkamendje, 2020-07-04:
Closing the issue since there is already a pending solution.

commit a7e6139b (2020-07-06): Bump OpenROAD commit

Fixes #104
Fixes pin not covered by guide error on bp_fe
closed 2020-07-04 by gkamendje