The-OpenROAD-Project/OpenROAD-flow #113: missing docker command

noytzach, 2020-07-12:
I followed the steps according to [this readthedocs page](https://openroad.readthedocs.io/en/latest/user/getting-started.html#option-2-build-from-sources) (build from sources with docker).

When I got to ```Verify installation```, nothing seemed to work or be installed.

I finally found [this page](https://github.com/The-OpenROAD-Project/OpenROAD-flow/blob/master/README.md) which tells you to

> Start an interactive shell in a docker container using your user credentials

this step is missing from [this readthedocs page](https://openroad.readthedocs.io/en/latest/user/getting-started.html#option-2-build-from-sources).

rovinski, 2020-07-12:
Documentation being out-of-date is a known issue. We don't have the resources to fix it in the near-term. The in-repo documentation is your best bet.

noytzach, 2020-07-13:
OK.

I used the **_User-Guide / Getting Started_** link from the homepage of [The OpenRoad Project](https://theopenroadproject.org/).
Maybe you should update it to point to the in-repo README.md

Regards